wltechblog/thingino-installers #5: Wyze Cam3 only blue light

JochenKlenk, 2024-11-08:
Hi,

I moved from my old RTSP Firmware to latest supported 4.36.9.139.
Then I check with the flasher tool the firmware which I needed and that is.
thingino-wyze_c3_t31al_atbm

After I copied that firmware file also th ethe sd card and power up the cam again it just shows blue light.
No AP network shows up. No reaction or anything else.
If I repower it it just need a sec and then again the blue light.

How I get further or how I get back to wyze orginal firmware?

Thanks a lot

wltechblog, 2024-11-08:
Several possibilities as to what went wrong, easiest path for support is to jump on our Discord and we'll get you sorted out.

wltechblog, 2024-12-16:
Has your issue been solved?

wltechblog, 2025-03-14:
Closing for inactivity

closed 2025-03-14 by wltechblog