wltechblog/thingino-installers #8: Wyze Cam Pan (v1) support

brandoningli, 2024-12-16:
Based on what I can find about firmware installation for the pan, it should be very similar to the Wyze Cam v2, just with different firmware. For reference, the instructions for installing their [webcam firmware](https://support.wyze.com/hc/en-us/articles/360041605111-Webcam-Firmware-Instructions) are the exact same between the two.

Thanks for all that you and the thingino team are doing to keep these things alive and useful!

wltechblog, 2024-12-16:
Indeed it is a very similar device, I have one on my bench and will be doing an installer + video very soon! I don't publish anything until I have tested on my own devices.

Tobe2d, 2025-01-12:
Hopefully this will support Cam Pan v3 too

wltechblog, 2025-01-12:
Cam Pan V3 won't be supported as they use SecureBoot to prevent loading third party firmware.

li20034, 2025-02-08:
Will a Cam Pan V2 work by following the Cam V3 instructions (except replacing the firmware image)? Or does that need a different bootloader? Thanks.

wltechblog, 2025-03-14:
I have a Pan V1 installer that is working well, and a video walk-through on my channel showing how to use it!

closed 2025-03-14 by wltechblog