wltechblog/thingino-installers #10: V3 Blue LED ONLY

edge360, 2024-12-21:
Experiencing the same issue as another user, didn't want to hijack their thread.  Just blue light no AP. Can't connect via Wyze app, don't see it on wifi. Popped SD card back into computer and same files were there, no log.  Any suggestions? 

vendor: 0x007a
device: 0x6011
t31x

Was trying to follow thingino instructions prior, but was unsure about which version so I went this route.

[T31X, GC2053, ATBM6031, 16MB](https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_c3_t31x_atbm.bin)
[T31X, GC2053, RTL8189FTV, 16MB](https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_c3_t31x_rtl.bin)

Can pop into the discord when I have time.

wltechblog, 2024-12-25:
If you don't have any new files written then no changes should have been made to the cam. Jump on Discord for sure for help!

wltechblog, 2025-03-14:
Closing for inactivity!

closed 2025-03-14 by wltechblog