wltechblog/thingino-installers #13: Can't open the latest zip file for wyzecam v2

rpcribari, 2025-02-11:
I just tried downloading the new zip file but it says its invalid when I try to open it.

rpcribari, 2025-02-12:
> Can you verify the md5sum of the zip file?
> 07371f103731043a3e39a67dc5260047 wyze-cam-2.zip
>
>
>
> Can you verify the md5sum of the zip file?
> 07371f103731043a3e39a67dc5260047 wyze-cam-2.zip

They are not the same. I've redownloaded multiple times. Still no luck.

Ryan P. Cribari
Cribari Computer Solutions LLC.
(303)503-3628
***@***.***
www.helpmeryan.com

wltechblog, 2025-02-12:
Are you doing a right click->save as from the file list? if so you're downloading a html document. The direct link to the zip file is https://github.com/wltechblog/thingino-installers/raw/refs/heads/main/wyze-cam-2/wyze-cam-2.zip

rpcribari, 2025-02-12:
That is exactly what I was doing wrong. Firmware installed. Thank you! 👍

Ryan P. Cribari
Cribari Computer Solutions LLC.
(303)503-3628
***@***.***
www.helpmeryan.com

closed 2025-02-12 by wltechblog