wltechblog/thingino-installers #15: Wyze Pan V1 - No video

apristel, 2025-03-13:
Flashed and updated fine.

Webui shows no video ( broken cam image )

I reset the camera via the UI and same result.

Camera was fine with Wyze firmware.

Any ideas?

wltechblog, 2025-03-13:
in the web ui, go to Information -> Share Diagnostic Info, send me the resulting URL.

apristel, 2025-03-13:
> in the web ui, go to Information -> Share Diagnostic Info, send me the resulting URL.

I did that and the link never appears?  It has internet access as it was able to grab the time from the time servers fine.

[image]

apristel, 2025-03-13:
Well, I unplugged it to give it a fresh boot and now it is completely bricked - does absolutely nothing.  Great.

apristel, 2025-03-13:
I opened the camera up and see the jtag pins - is there a way to force a reflash?

wltechblog, 2025-03-13:
There is a UART but not a JTAG. If the device is actually bricked it cab be recovered with Cloner. I'll do a video on unbricking the Pan V1 today, I'll ping this ticket when its live.

wltechblog, 2025-03-13:
Unbrick video is live: https://youtu.be/tdl5S-dIbwI

apristel, 2025-03-13:
Thanks!  I was trying prior to the video - never thought i'd be using that port on the camera.

[image]

I'm getting this in cloner.   Also the windows version doesn't that the ddr full whatever it was ( video isn't in front of me) but has these...

[image]

apristel, 2025-03-13:
oops  these options

[image]

wltechblog, 2025-03-13:
you want the t20_sfc_nor_writer_full.cfg

and on the policy page you want just one policy of full and select the thingino firmware file.

apristel, 2025-03-13:
Getting closer!

Any thoughts?  And I want to tell you thanks again.  

[image]

wltechblog, 2025-03-13:
your policy page should look like this 

[image]

apristel, 2025-03-14:
I feel like this thing wants to eat the concrete! lol.

[image]

apristel, 2025-03-14:
WAIT!!   I just made a new cable and ..

[image]

YOU THE MAN!  THANKS!!!!!!! :)

wltechblog, 2025-03-14:
Awesome!

apristel, 2025-03-14:
Well, we have a working firmware, however, still no video.

The generate link does not produce said link.

wltechblog, 2025-03-14:
On the menu go to Information -> Commands and copy the dmesg info, ant attach it here.

apristel, 2025-03-14:
0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14 (buildroot@buildroot) (buildroot-gcc-14.2.0) #1 PREEMPT Sun Dec 8 10:16:40 UTC 2024
[    0.000000] CPU0 RESET ERROR PC:8A401DD3
[    0.000000] CPU0 revision is: 00d00101 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] CCLK:860MHz L2CLK:430Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 0046e000 @ 00010000 (usable)
[    0.000000]  memory: 00032000 @ 0047e000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 05e00000 @ 00000000 (usable)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x05dfffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x05dfffff]
[    0.000000] On node 0 totalpages: 24064
[    0.000000] free_area_init_node: node 0, pgdat 80476bc0, node_mem_map 81000000
[    0.000000]   Normal zone: 188 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 24064 pages, LIFO batch:3
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00101],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 23876
[    0.000000] Kernel command line: mem=94M@0x0 rmem=22M@0x5E00000 console=ttyS1,115200n8 panic=10 root=/dev/mtdblock4 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),32k(env),224k(config),1504k(kernel),5824k(rootfs),-(rootfs_data),15872k@0x80000(upgrade),16384k@0(all)
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Memory: 89888k/96256k available (3773k kernel code, 6368k reserved, 762k data, 200k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:418
[    0.000000] clockevents_config_and_register success.
[    0.000020] Calibrating delay loop... 858.52 BogoMIPS (lpj=4292608)
[    0.090048] pid_max: default: 32768 minimum: 301
[    0.090315] Mount-cache hash table entries: 512
[    0.090882] Initializing cgroup subsys debug
[    0.090906] Initializing cgroup subsys freezer
[    0.093156] devtmpfs: initialized
[    0.095338] regulator-dummy: no parameters
[    0.095704] NET: Registered protocol family 16
[    0.118646] bio: create slab <bio-0> at 0
[    0.126840] jz-dma jz-dma: JZ SoC DMA initialized
[    0.128492] usbcore: registered new interface driver usbfs
[    0.128748] usbcore: registered new interface driver hub
[    0.129045] usbcore: registered new device driver usb
[    0.129517]  (null): set:249  hold:250 dev=100000000 h=500 l=500
[    0.131168] media: Linux media interface: v0.10
[    0.131466] Linux video capture interface: v2.00
[    0.134288] cfg80211: Calling CRDA to update world regulatory domain
[    0.135368] Switching to clocksource jz_clocksource
[    0.138834] dwc2 otg probe start
[    0.138873] jz-dwc2 jz-dwc2: cgu clk gate get error
[    0.138902] DWC IN OTG MODE
[    0.139756] dwc2 dwc2: Keep PHY ON
[    0.139773] dwc2 dwc2: Using Buffer DMA mode
[    0.139788] dwc2 dwc2: Core Release: 3.00a
[    0.139837] dwc2 dwc2: DesignWare USB2.0 High-Speed Host Controller
[    0.139878] dwc2 dwc2: new USB bus registered, assigned bus number 1
[    0.142010] hub 1-0:1.0: USB hub found
[    0.142057] hub 1-0:1.0: 1 port detected
[    0.142214] dwc2 dwc2: DWC2 Host Initialized
[    0.142738] dwc2 dwc2: enter dwc2_gadget_plug_change:2588: plugin = 1 pullup_on = 0 suspend = 0
[    0.142764] dwc2 otg probe success
[    0.143120] NET: Registered protocol family 2
[... 393 lines cut ...]
[   12.659367] RTW: rtw_cfg80211_set_encryption, ret=0
[   12.659742] RTW: rtw_ndev_notifier_call(wlan0) state:4
[   12.659862] RTW: cfg80211_rtw_change_station(wlan0) mac:5c:e9:31:74:76:7c
[   12.660366] RTW: set pairwise key camid:0, addr:5c:e9:31:74:76:7c, kid:0, type:AES
[   12.660726] RTW: set group key camid:1, addr:5c:e9:31:74:76:7c, kid:1, type:AES
[   13.182949] RTW: OnAction_back
[   13.182969] RTW: OnAction_back, action=0
[   13.184719] RTW: issue_addba_rsp_wait_ack(wlan0) ra=5c:e9:31:74:76:7c status:=0 tid=0 size:16, acked, 1/3 in 0 ms
[   14.525541] RTW: Turbo EDCA =0x5ea42b
[   22.525546] RTW: Turbo EDCA =0xa42b
[   24.215897] RTW: OnAction_back
[   24.215917] RTW: OnAction_back, action=0
[   24.220207] RTW: issue_addba_rsp_wait_ack(wlan0) ra=5c:e9:31:74:76:7c status:=0 tid=1 size:16, acked, 1/3 in 0 ms
[   24.525797] RTW: Turbo EDCA =0x5ea42b
[   25.569742] codec_codec_ctl: set repaly channel...
[   25.569772] cgu_set_rate, parent = 860160000, rate = 4096000, n = 210, reg val = 0x200020d2
[   25.569783] codec_codec_ctl: set sample rate...
[   25.569905] codec_codec_ctl: set device...
[   25.805461] codec_set_device: set device: speaker...
[   26.647106] set sensor gpio as PA-low-10bit
[   26.672716] jxf22 0-0040: chip found @ 0x40 (i2c0) is not an jxf22 chip.
[   28.525673] RTW: Turbo EDCA =0xa42b
[   30.525542] RTW: Turbo EDCA =0x5ea42b
[   34.525557] RTW: Turbo EDCA =0xa42b
[   36.529355] RTW: Turbo EDCA =0x5ea42b

wltechblog, 2025-03-14:
[ 26.672716] jxf22 0-0040: chip found @ 0x40 (i2c0) is not an jxf22 chip.

well well well

you have a different hardware config than we've seen before. It's probably a jxf23, but if you can ssh in and run a couple commands we can verify.
Paste these in:
> /etc/init.d/S95prudynt stop
> rmmod sensor_jxf22_t20;i2cget -y  0 0x40 0x0a w

wltechblog, 2025-03-14:
I'll pick this back up tomorrow, day is over here.

apristel, 2025-03-14:
No worries! Appreciate you.

If you are still there is the ssh pass the same as the root one we set?
Assuming it is.

wltechblog, 2025-03-14:
Yes its the same password.

apristel, 2025-03-14:
[image]

wltechblog, 2025-03-14:
Try this fresh build: https://wanderlounge.net/random/thingino-wyze_cp1_jxf23.bin

Rename the file to autoupdate-full.bin and put it on the SD card. Make sure there isn't an autoupdate-full.done on the card, and boot the cam with the card inserted. The flash takes 3-4 minutes and you should see the config wifi portal again.

apristel, 2025-03-14:
Flashed and no video yet, however the logs link works now..
http://tb.thingino.com/djzz

wltechblog, 2025-03-14:
I still don't see confirmation of the sensor. Do me a favor and follow this guide for loading sinfo and come back with its results please!

https://github.com/themactep/ingenic-sdk/releases/tag/sinfo

apristel, 2025-03-14:
What is the best way to get the module on to the camera?  I tried scp,
sftp....

wltechblog, 2025-03-14:
scp should work fine but you can also copy it to the sd card and load it from /mnt/mmcblk0p1/

apristel, 2025-03-14:
OK, the t20 module correct?

Thanks again!

wltechblog, 2025-03-14:
Yes T20

apristel, 2025-03-14:
root@wyze-kitchen ~# cd /mnt/mmcblk0p1
root@wyze-kitchen mmcblk0p1# insmod /mnt/mmcblk0p1/sinfo-t20.ko
root@wyze-kitchen mmcblk0p1# echo 1 > /proc/jz/sinfo/info
root@wyze-kitchen mmcblk0p1# dmesg | tail
[  159.176075] RTW: Turbo EDCA =0xa42b
[  159.225629] sinfo: [Error] Function: sensor_read, Line: 262, I2C transfer failed with return code: -5
[  159.225645] sinfo: Read from address: 0x3b00, Value: 0x0
[  159.225656] sinfo: [Error] Failed to read sensor at address 0x3b00, value read: 0x0
[  159.225682] wait stable.[247][cgu_cim]
[  159.315848] sinfo: Read from address: 0xa, Value: 0xf
[  159.316262] sinfo: Read from address: 0xb, Value: 0x22
[  159.316283] sinfo: Successful sensor detection: jxf22, I2C Bus: 0, I2C Address: 0x40
[  161.180883] RTW: Turbo EDCA =0x5ea42b
[  163.180944] RTW: Turbo EDCA =0xa42b

wltechblog, 2025-03-14:
No don't use that one, grab the latest: https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_cp1.bin

apristel, 2025-03-14:
Well..

[image]

I don't get it and I feel terrible wasting your time.  I flashed all my other cameras without issue!  

Really appreciate your time and I will be sending you something for your troubles!

apristel, 2025-03-14:
> So we're working with the new build?

I did not receive the link for the build a few posts up in time - so I just
grabbed the files like I did the first time and sent it to the card via
rufus - and popped the card in and powered the camera up - setup the
pwd/wifi/etc and viola.

I did that same procedure several times prior to ever starting an issue.. I
don't understand what changed!

wltechblog, 2025-03-14:
Is it possible the ribbon cable was knocked wonky? that could do it

apristel, 2025-03-14:
It is possible, however the first time I didn't have it ripped apart, I
only took it apart to do the cloner stuff.

apristel, 2025-03-14:
Where is a donate button!?  Or Vemno?

wltechblog, 2025-03-14:
if I ever figure out perks I'll be adding memberships to my channel but for now... feel free to check out my cams and gadgets lists on amazon, they're on the channel description :)

wltechblog, 2025-03-14:
Closing this issue!

closed 2025-03-14 by wltechblog