wltechblog/thingino-installers #21: Automated installer generation

yebo29, 2025-04-23:
Would it make sense to somehow automate installer build/generation on a weekly basis? Haven't looked at details yet because I wanted to poll here first. Thinking perhaps some sort of git workflow/pipeline?

wltechblog, 2025-04-23:
it certainly could be automated however I try to be a little bit scrutinizing about the firmware we bake in and have the expectation that the firmware works... it's not a perfect system but without some sort of formal automated testing I'm just not comfortable with the yolo method.

thompa, 2025-04-28:
Using 'wyze-cam-2-sd and Balena Etcher to burn the sd.img file to an SD card and came up with an error:(0,h.requestMetadata) is not a function.