wltechblog/thingino-installers #22: Support is gatekept in propriatary software

russ-go, 2025-05-11:
Discord is a serious weakspot in any opensource project that relies on it for any support or knowledge base.

Project should accept that and move to a non-compromised project other than discord.  The free content hosting isn't worth it.

wltechblog, 2025-05-11:
The Thingino project is primarily hosted on Github, and we use Discord and Telegram for real time communicaiton between developers and users.  if you don't like one of them, pick one of the other two.

closed 2025-05-11 by wltechblog

gtxaspec, 2025-05-11:
> Discord is a serious weakspot in any opensource project that relies on it for any support or knowledge base.
> 
> Project should accept that and move to a non-compromised project other than discord.  The free content hosting isn't worth it.
Telegram is also available.