wltechblog/thingino-installers #25: Testing Assistance?

gregorydlogan, 2025-05-16:
I've got what Amazon calls an Pan V2, though the back says it's a 'DF3' in Chinese.  I see the installer is marked as not-ready, can I help?

wltechblog, 2025-05-18:
The Wyze Cam Pan V2 installer has been tested by a few people successfully, I just don't have one myself to validate with.  I'm not sure about your DF3 marking, is that instead of the normal Wyze sticker?

gregorydlogan, 2025-05-19:
You know what, I've actually got two devices - one a legit Wyze Pan v2 with the correct sticker and everything, and then this DF3 with a non-Wyze sticker and back grill.

It's otherwise visually identical to a Pan, and I've had wz_mini and others applied to it in the past so I know it doesn't have a locked bootloader at least! Will try and get wz_mini working on it again to gather more hardware info.