wltechblog/thingino-installers #30: update - lights disabled

sn0n, 2025-06-14:
forgot the old password i had setup on the camera, and trying to do the update and *hopefully* reset it,...

however, my device shows no lights. so not sure what to do exactly.. suggestions?

wyze vam v2

sn0n, 2025-06-14:
unrelated, the main page for v2 has a small typo : https://github.com/wltechblog/thingino-installers/tree/main/wyze-cam-2
 "tiem" in the first sentence. ^>^

citizen4013, 2025-07-13:
Not sure about the light being disabled but flashing the Neos 2 seems to have bricked the camera. No light, nothing.

wltechblog, 2025-07-14:
> Not sure about the light being disabled but flashing the Neos 2 seems to have bricked the camera. No light, nothing.

Check out my unbrick video for that cam, it's straightforward.